UofSSpaceTeam/rover-hardware #17: TUSB2077A USB hub

ottopasuuna, 2016-12-02:
Make a usb hub based on the TUSB2077A chip by TI.
- [ ] remove over current protection (for simplicity)
- [ ] Invesigate/remove resistor network
- [ ] Simplify connector circuits
- [ ] Redo layout

ausshir, 2016-12-02:
Should we make sure to use through-hole USB A plugs? The reference design seems To use SMT but I'd be afaird of the big plug tearing off the pads.

ottopasuuna, 2016-12-02:
Yeah, that sounds like a good idea.

ottopasuuna, 2017-01-19:
This is basically done. It needs a proper reset circuit; either a power-on reset chip, or an rc circuit that properly discharges after a sudden power down. All the ports seem to work, and the rover can communicate to VESCs plugged into it without issues.

closed 2017-01-19 by ottopasuuna