UofSSpaceTeam/rover-hardware #35: Replace USB hub for RasPi's

tyrelkostyk, 2017-12-06:
Pretty much all four of the mounting holes are broken on the USB hub, so the entire board is gonna need replacing. Needed in time for the completion of the Electrical Tray design, **Early January 2018**

ottopasuuna, 2017-12-06:
Alternatively we could make our own USB hub integrated into the power board for the Pi's. Would need to have 7 ports, and probably USB 2.0, as 1.0 or full speed USB brings down the data rate of the entire bus to 12 Mbit/sec if I recall correctly. I made a 1.0 hub last year and the layout took about a week while classes were going on, so it wouldn't be an unreasonable thing to do.

tyrelkostyk, 2017-12-07:
Current design plan was to try have the 4 raspi's, the USB hub, & the Pi power board all mounted together, and therefore would need to be relatively similar in shape. Would this size constraint be possible with your idea?

ottopasuuna, 2017-12-07:
It should be possible. Power distribution shouldn't take much space, and you can use both sides of the board. Probably good to start with just the power, and see if it possible to add USB later. We'll have difficulty purchasing a USB hub with mounting holes that line up with a Raspberry Pi, so we'll either need a mechanical adapter or build something custom anyway.

ottopasuuna, 2017-12-07:
Also remember that you need an ethernet switch for the Pi's as well, doesn't have to be mounted on the stack of Pi's, but may be convenient if it did.